issalig/toshiba_air_cond #14: older model with 3 wires

olivluca, 2022-07-24:
Hello, I have an older model (around 30 years old) RAV-260BH/260AH8,  where the remote uses 3 wires: two for power supply (12 or 15v AC) and the remaining one for bidirectional communication. I don't have an oscilloscope, but since it uses a photo thyristor for sending and an optocoupler for receiving, I suppose it uses half of the ac wave for sending and the other half for receiving at 50bps.
This is what the service manual says
[image]
but it has no other details about the protocol used. 
Finding information on this old unit is quite difficult, per chance do you happen to know something about it?

issalig, 2022-07-25:
Hi, I have no information about it. :(
The best way to guess what's happening is to use an oscilloscope. As you do not have one, maybe you can try with an arduino/esp8266 and the correspondent resistor divider (to convert voltages to to 0-5v for atmega or 0-3.3v for esp). Then with serial plotter from teh arduino IDE you can draw the values.
Also a logic analyzer would work, here it is an example https://aliexpress.com/item/1005003812410259.html

olivluca, 2022-07-25:
Thank you, if I weren't too scared to break the remote (since the spares are now unobtanium) I'd go with the plan b. I'll see if I can borrow an oscilloscope.

closed 2022-07-25 by olivluca