issalig/toshiba_air_cond #27: Send temperature from remote controller

hamishfagg, 2023-02-07:
Hi there,
Is your code able to send a temperature measurement from the remote (or from your PCB/wifi) to the indoor unit? I've had a look through the code and I don't think it's there but I might be missing it. Would it be hard to add?

issalig, 2023-02-07:
What do you mean? If it its possible to set a target temperature? If that's the question, the answer is yes aand it is done by function air_set_temp at
https://github.com/issalig/toshiba_air_cond/blob/9bd722440544486d960be91356f493cc87f417a3/air/toshiba_serial.hpp#L136

issalig, 2023-02-07:
Well, re-reading your message maybe you mean the status messages that remote sends to indoor unit like this one:
```
40 00 55 05 08 81 00 7C 00 E5 (heat mode 24)

40 00 55 05 08 81 00 68 00 f1 
      |           |  |- 0110 1000 -> 104/2 -> 52 - 35 = 17   (temp)
      |-opc1         |-bit0 ON:1 OFF:0                 
 ```   

In such a case, you are true I have not function to build these messages. I just decode them because I used remote and my pcb together. A function for sending temp status would be like. Of course without wired controller you would need an external thermometer to read the date and set it to sensor_temp variable.
```
void air_send_status(air_status_t *air, uint8_t value)  {
  //             00    01    02    03    04    05    06    07    08    09    10    11   CRC
  byte data[] = {0x40, 0x00, 0x55, 0x05, 0x08, 0x81, 0x00, 0x7C, 0x00, 0xE5}; //dummy msg

  //send sensor temp
  data[7] = ((air->sensor_temp) + 35) << 1; //temp is bit7-bit1
  data[7] = data[7] + air->power; //set bit 0

  //compose new message
  //compute crc
  data[9] = XORChecksum8(data, sizeof(data) - 1);

  air_send_data(air, data, sizeof(data));
}
```

issalig, 2023-02-08:
If there is not remote and the pcb is alone, it would be necessary to have a ping function

```
void air_send_ping(air_status_t *air, uint8_t value)  {
  //             00    01    02    03    04    05    06    07    08   09   10    CRC
  byte data[] = {0x40, 0x00, 0x15, 0x07, 0x08, 0x0C, 0x81, 0x00, 0x00, 0x48, 0x00, 0x9F}; //dummy msg

  //compose new message
  //compute crc
  data[11] = XORChecksum8(data, sizeof(data) - 1);

  air_send_data(air, data, sizeof(data));
}
```

hamishfagg, 2023-02-09:
Thank you for all of your tips on this! I will have a go at it when I have some time.

hamishfagg, 2023-04-11:
I've discovered when trying to install my PCB that my heatpump is missing several components on the AB terminal board (most notably a transformer) so unfortunately I can't install it. But thanks for your help!

closed 2023-04-11 by hamishfagg

issalig, 2023-04-11:
Ooops. Could you post a photo? Maybe there is an easy fix.