issalig/toshiba_air_cond #28: toshiba_air_cond instead remote controller?

MrTheBarbarian, 2023-02-15:
Hello! 
Great job you have done!
Is it possible to use this device instead remote controller?
I have ordered Estia R32 heat pump, but I didn't order remote controller. I thought is it possible to control the heat pump remotely with this device if I do not have remote controller.
Thanks!

issalig, 2023-02-15:
I have not tried it but I think it is possible.
Probably the only thing you need is to periodically report the temperature (you will need also a temperature sensor) with the function mentioned in this issue https://github.com/issalig/toshiba_air_cond/issues/27#issuecomment-1420507111

Titoflegit, 2025-01-09:
I made some measurements on the AB bus of my toshiba estia serie 3 heatpump.
Electrical levels are almost the same but a bit higher at 16.5 volts /14volts, baudrsye at 2400 but without parity. Then the messages are completely differrent there is a fixed frame without response each 5 seconds beginning with F0F0 and ending with A0. And if we change something on the controller this frame is longer but still ending with A0 so there is no more a CRC.

Titoflegit, 2025-01-09:
Estia Series 3. Controller HWS AMS11E
Periodic Frame without response each 5 seconds.
F0F00A70FFFF003AB2A0

Frame sended when temperature changed to 45 degrees:
F0F00F6070E00123027A668A7AC9A0
Acknowledged with:
F0F00A7060E080A1DBA0

Frame sended when temperature changed to 44 degrees:
F0F00F6070E00123027**8**668A7**8**C**5**A0 (changes in **BOLD** vs 45 degres)
Acknowledged with:
F0F00A7060E080A1DBA0 (same as 45 degrees)

Frame sended when temperature changed to 43 degrees:
F0F00F6070E00123027**6**668A7**6**C**1**A0 (changes in **BOLD** vs 45 degres)
Acknowledged with:
F0F00A7060E080A1DBA0 (same as 45 degrees)

There is also sometimes some other frames that are exchanged without pressing any button on the controller, so we cannot simply emulate the controller. I doubt also that the temperature change frame will be the same in the case of the addition of a second remote controller and don't want to spend 300EUR to make measurments.

From series 4 and 5 (and also R32 I suppose) there is a ABbus to Modbus interface available, at also 300EUR but it would be the simplest solution I think.

For me, I will go the plan B, i.e. optocoupler to act like the pushbutton is pressed, or less invasive plan B2 with some electromagnet pressing the buttons, but only practical for a few buttons, perhaps only the temperature)

Hope to have contributed. I you think that some scope screenshots can be usefull, jsut ask, the setup si still in place

Titoflegit, 2025-01-09:
Regarding your backup plan to emulate button pressure with optocouplers, how did you proceed to locate the vias to short with the optocoupler ?

issalig, 2025-01-09:
Visually follow the trace of the two parts of the button footprint, then stop at a via (where you can solder). Double check it with a tester (beep in diode checking mode).

issalig, 2025-01-09:
> Estia Series 3. Controller HWS AMS11E Periodic Frame without response each 5 seconds. F0F00A70FFFF003AB2A0
> 
> Frame sended when temperature changed to 45 degrees: F0F00F6070E00123027A668A7AC9A0 Acknowledged with: F0F00A7060E080A1DBA0

Great, did you manage to decode the protocol?
If not, I can take a look next week.

> Hope to have contributed. If you think that some scope screenshots can be usefull, jsut ask, the setup is still in place
Yes, feel free to add screenshots that document it.
Thanks!

issalig, 2025-01-09:
About the collision, I would not worry about it. Just send messages twice :)

issalig, 2025-01-10:
Some progress, F0F0  (1111 0000 1111 0000) and A0 (1010 0000) are normally used as start of frame and end of frame markers.
```
F0 F0 0F 60 70 E0 01 23 02 7A 66 8A 7A C9 A0
                                          |- end
                                       |- CRC
                                    |- temp*2 +32
                              |- 66 8A could be mode or speed
                           |- repeated temp?
               |- E0 is common byte
            |- dest 70 is master, FF is everybody?
         |- source  60 is remote
|- begin
```

Temperature seems to follow this rule.
protocol_value = (temperature × 2) + 32
45°C: (45 × 2) + 32 = 122 (0x7A)
Temperature is repeated, do not why

66 8A seem to be mode

CRC is used and is the last byte but one.
CRC is computes as the sum of the bytes without markers modulo 256

0F + 60 + 70 + E0 + 01 + 23 + 02 + 7A + 66 + 8A + 7A = 3C9
3C9 mod 256 = C9

Can you change the mode, fan speed, save, etc, ... and post the messages?

Titoflegit, 2025-01-10:
So situation at start is heating zone 1 active in autotemp mode, hot water disabled.
Following the frames are witjout the starting F0F0 and the endind A0
Sometimes as described before there is a status ?? sended. longer thant the ping ?? each 5 seconds, I name t a PeriodicLong.

So PeriodicLong:
0F6070E0410C8100004800D5
Turn off Autotemp mode, actual temp setting is 35°
0C6070E001240100E2
PeriodicLong
0F6070E0410C8100004800D5
Turn on Autotemp
0C6070E001240101E3
Turn off Zone heating
0B6070E0012902DF
Turn on Zone heating
0B6070E0012103E0
Turn On Hot water watertemp settings was 55°
0B6070E001210CE9
Water temp to 50°,(no trace)  then Hot water off
0B6070E06131004D
WATER ON
0B6070E001210CE9 (so temperature setting is not in the command)
Hot water temp down to 49°
OF6070E0012308666681669F
PeriodicLong
0F6070E410C8100004800D5
Hot water temp down to 48°
OF6070E00123086666A0669D
Hot water temp down to 47°
OF6070E001230866667E669B
Periodic:Long
0F6070E0410C8100004800D5
Hot Water OFF
0B6070E0012108E5
Night Mode ON
0C6070e01324880871
Night Mode OFF
0C6070E00124A80069
Now we are back in autotemp mode as at the start.

issalig, 2025-01-10:
```
Opcode 0C

F0 F0 0C 60 70 E0 01 24 01 01 E3 A0
                           |- 0 OFF 1 ON  Autotemp

Opcode 0B

F0 F0 0B 60 70 E0 01 21 08 E5 A0
                        |- 0C (1010) ON 08 (1000) OFF  Hot water  -> bit 1 and bit 3?
F0 F0 0B 60 70 E0 01 21 03 E0 A0
                        |- 3 (0011) ON 2 (0010)  OFF  Zone heating  bit0,   bit1 is always set

Opcode 0F 
F0 F0 0F 60 70 E0 01 23 08 66 66 81 66 9F A0
                                 |-(129-32)/2 = 48.5 water temp ???
F0 F0 0F 60 70 E0 01 23 08 66 66 7E 66 9B A0
                                 |-(126-32)/2 = 47 water temp
                              |-(0x66-32)/2 = 35 current temp setting
                                 
We already know
F0 F0 0F 60 70 E0 01 23 02 7A 66 8A 7A C9 A0
__ __ __ __ __ __ ?? ?? ?? __ ?? ?? __ __ __
                                          |- end
                                       |- CRC
                                    |- temp*2 +32
                              |- 66 8A could be mode or speed?/ another temp?
                           |- repeated temp?
               |- E0 is common byte
            |- dest 70 is master, FF is everybody?
         |- source  60 is remote
       |- length
|- begin
    
```
So we add

Titoflegit, 2025-01-11:
F0 F0 0F 60 70 E0 01 23 02 7A 66 8A 7A C9 A0
__ __ __ __ __ __ ?? ?? ?? __ ?? ?? __ __ __
                                          |- end
                                       |- CRC
                                    |- temp*2 +32
                              |- 66 8A could be mode or speed?/ another temp?
                           |- repeated temp?
               |- E0 is common byte
            |- dest 70 is master, FF is everybody?
         |- source  60 is remote
       |- length
|- begin

8A is perhaps hot water temperature 53 degrees

issalig, 2025-01-11:
To sum up.

```
Periodic message from remote to master, status?
F0 F0 0F 60 70 E0 41 0C 81 00 00 48 00 D5 A0

F0 F0 0A 70 FF FF 00 3A B2 A0 //pulse every 5s  master to all
F0 F0 0A 70 60 E0 80 A1 DB A0 //ack from master  

-------
Autotemp
F0 F0 0C 60 70 E0 01 24 01 01 E3 A0
                           |- 0 OFF 1 ON  Autotemp

-------
Hot water
F0 F0 0B 60 70 E0 01 21 08 E5 A0
                        |- 0C (1010) ON 08 (1000) OFF  Hot water  -> bit 1,  bit 3 is always set ??

Zone heating
F0 F0 0B 60 70 E0 01 21 03 E0 A0
                        |- 3 (0011) ON 2 (0010)  OFF  Zone heating  bit0,   bit1 is always set ??

-------
F0 F0 0F 60 70 E0 01 23 02 7A 66 8A 7A C9 A0
__ __ __ __ __ __ ?? ?? ?? __ __ __ __ __ __
                                          |- end
                                       |- CRC sum of bytes without (F0F0 and CRC A0) % 256 
                                    |- temp*2 +32  7A -> 45º
                                 |- 8A -> 53º water temp
                              |- 66 -> 35º other temp
                           |- 8A -> 53º repeated temp after 3 bytes?
                   |- 01 23 02 unknown                           
               |- E0 is common byte
            |- dest 70 is master, FF is everybody?
         |- source  60 is remote
       |- length
|- begin
    
```

issalig, 2025-01-19:
Any progress? @Titoflegit

Titoflegit, 2025-01-19:
I wait on the usb to serial TTL converter delivery so I will be able to sniff with my laptop and make some trials to send frames and see if they are accepted. Will keep you informed

issalig, 2025-01-19:
What are you going to use for communication? My circuit and a ttl to usb?

Titoflegit, 2025-01-19:
Yep. Without the optocoupler. I did not find the way to interface correctly to the rs485 usb converter I have already

issalig, 2025-01-20:
you can use a simple usb ttl adapter https://aliexpress.com/item/32637261853.html which I think is the one you are waiting for.

Also  you can try an ESP8266, or even an arduino

Titoflegit, 2025-01-30:
So I use your circuit and a ttl usb converter, I engraved a little pcb on my cnc milling maching and soldered it yesterday.
  
But there is problem, the rx line is in the wrong polarity, normally idle is high and with your circuit it is not rhe case. Did I miss something?
  
I will correct it by cabling the opto toward gnd rather than vcc.
  
Hope it will work.

issalig, 2025-01-30:
Hmm, if you disconnect AB wires you still get "1" ?

Titoflegit, 2025-01-30:
Yesterday I got idle voltage on rx at around 2v and the datastream as positive pulses to 3v3.
  
But I agree that it is not according your schematics where the opto light when voltage is zener + Vf doode + Vf opto ..
  
I will check tje voltage threshold with a lab voltage source

issalig, 2025-01-30:
As a dirty trick you can add one more diode such as 1n4001, to drop another 0.7v

I am not an electronics expert, just self-learner, I have seen others using opamps and that is probably a better solution than the zener workaround.

Titoflegit, 2025-01-30:
In fact there was 2 problems the threshold that I solved by changing the zmm13 by a 5v6 and 8v2 in series and also the low level volrage because the usbttl adapter has a led directly on the rx so your 1k was too high to have a valid low signal...
  
Nevertheless your design has the best advantage Simplicity!!

Titoflegit, 2025-01-30:
Ok it works and my python script sniff it. With SOF end EOF removed.

This is doing nothing on the remote control, mode is zone 1 in automatic 
mode, DHW off

What are your suggestions for the next steps?

-----------------------------------

Starting new event loop
main sniffer task started
2025-01-30 16:42:07,476 Serial Port is now opened
2025-01-30 16:42:07,476 Opening serial interface:
  port: COM3
  baudrate: 2400
  bytesize: 8
  parity: none
  stopbits: 1
  timeout: 0.01375

2025-01-30 16:42:08,056 Frame wo SOF and EOF: 0B 60 70 E0 61 31 00 4D
2025-01-30 16:42:08,150 Frame wo SOF and EOF: 15 70 FF E0 A0 31 C1 04 12 
7E 66 66 80 6B 38 00 00 79
2025-01-30 16:42:11,766 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:16,949 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:22,131 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:27,313 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:32,399 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:35,088 Frame wo SOF and EOF: 0F 60 70 E0 41 0C 81 00 00 
48 00 D5
2025-01-30 16:42:35,150 Frame wo SOF and EOF: 10 70 60 E0 80 0C 00 08 00 
00 48 00 9C
2025-01-30 16:42:37,577 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:38,047 Frame wo SOF and EOF: 0B 60 70 E0 61 31 00 4D
2025-01-30 16:42:38,141 Frame wo SOF and EOF: 15 70 FF E0 A0 31 C1 04 12 
7E 66 66 80 6D 38 00 00 7B
2025-01-30 16:42:42,770 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:47,952 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:53,130 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:42:58,311 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:03,491 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:05,071 Frame wo SOF and EOF: 0F 60 70 E0 41 0C 81 00 00 
48 00 D5
2025-01-30 16:43:05,149 Frame wo SOF and EOF: 10 70 60 E0 80 0C 00 08 00 
00 48 00 9C
2025-01-30 16:43:08,077 Frame wo SOF and EOF: 0B 60 70 E0 61 31 00 4D
2025-01-30 16:43:08,187 Frame wo SOF and EOF: 15 70 FF E0 A0 31 C1 04 12 
7E 66 66 80 6E 39 00 00 7D
2025-01-30 16:43:08,672 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:13,852 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:19,136 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:24,313 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:29,496 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:34,672 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:35,016 Frame wo SOF and EOF: 0F 60 70 E0 41 0C 81 00 00 
48 00 D5
2025-01-30 16:43:35,094 Frame wo SOF and EOF: 10 70 60 E0 80 0C 00 08 00 
00 48 00 9C
2025-01-30 16:43:38,035 Frame wo SOF and EOF: 0B 60 70 E0 61 31 00 4D
2025-01-30 16:43:38,130 Frame wo SOF and EOF: 15 70 FF E0 A0 31 C1 04 12 
7E 66 66 80 6F 39 00 00 7E
2025-01-30 16:43:39,804 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:45,088 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:50,268 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:43:55,447 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:44:00,625 Frame wo SOF and EOF: 0A 70 FF FF 00 3A B2
2025-01-30 16:44:05,099 Frame wo SOF and EOF: 0F 60 70 E0 41 0C 81 00 00 
48 00 D5
2025-01-30 16:44:05,177 Frame wo SOF and EOF: 10 70 60 E0 80 0C 00 08 00 
00 48 00 9C
[... 326 lines cut ...]
target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You are=
 receiving this because you were mentioned.<img src=3D"cid:1738252424679ba0=
=3D"color: transparent; font-size: 0; display: none; visibility: hidden; ov=
erflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: =
/2624420649</span><span>@</span><span>github</span><span>.</span><span>com&=
gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_77337a9547a669000ba347e393e0f17c
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_77337a9547a669000ba347e393e0f17c--

--=_497f155b4c2939207a1d4b3baae85546--

issalig, 2025-01-30:
I would just record the power on/ off
and the send it to check if tx does work.

Or water on/off you previously decoded.

Titoflegit, 2025-01-31:
IT WORKS !!

In order to avoid collision, as I have seen on the records I made that 
the other frames are never?? sent immediately after a periodic each 5 
seconds, so when I detect the periodic, it simply activate a flag and if 
there is some commands in the pipeline, it send it.

Astonishingly, the remote controller see the acknowledge and change his 
display accorinding the command sent !!

Needs more testing but seems to work

BIG THANKS for your help!

======================================== RESTART: 
C:\Users\titof\Desktop\pyscripts\SerialSniffer\ABBusSniffer.py 
========================================
Starting new event loop
main sniffer task started
2025-01-31 17:24:24,233 Serial Port is now opened
2025-01-31 17:24:24,249 Opening serial interface:
  port: COM3
  baudrate: 2400
  bytesize: 8
  parity: none
  stopbits: 1
  timeout: 0.01375

Connected to mqtt broker
2025-01-31 17:24:25,392 Frame: 0A 70 FF FF 00 3A B2
Subscribed
2025-01-31 17:24:30,579 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:24:35,851 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:24:36,884 Frame: 0F 60 70 E0 41 0C 81 00 00 48 00 D5
2025-01-31 17:24:36,963 Frame: 10 70 60 E0 80 0C 00 08 00 00 48 00 9C
2025-01-31 17:24:39,888 Frame: 0B 60 70 E0 61 31 00 4D
2025-01-31 17:24:39,982 Frame: 15 70 FF E0 A0 31 C1 00 12 7E 84 66 6E 79 
3A 00 00 91
2025-01-31 17:24:41,030 Frame: 0A 70 FF FF 00 3A B2
MQTT Topic Received:  PAC/HeatTemp   b'47'
2025-01-31 17:24:46,300 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:24:46,316 -------------> COMMAND: F0 F0 0F 60 70 E0 01 23 
02 7E 66 66 7E AD A0
2025-01-31 17:24:46,425 Frame: 0F 60 70 E0 01 23 02 7E 66 66 7E AD
2025-01-31 17:24:46,472 Frame: 0A 70 60 E0 80 A1 DB
2025-01-31 17:24:46,831 Frame: 11 70 FF E0 C0 31 C1 00 12 7E 7E 66 00 86
2025-01-31 17:24:51,450 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:24:56,723 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:25:01,924 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:25:06,929 Frame: 0F 60 70 E0 41 0C 81 00 00 48 00 D5
2025-01-31 17:25:07,008 Frame: 10 70 60 E0 80 0C 00 08 00 00 48 00 9C
2025-01-31 17:25:07,134 Frame: 0A 70 FF FF 00 3A B2
MQTT Topic Received:  PAC/AutoMode   b'1'
2025-01-31 17:25:09,889 Frame: 0B 60 70 E0 61 31 00 4D
2025-01-31 17:25:09,889 Frame: 15 70 FF E0 A0 31 C1 00 12 7E 7E 66 6E 7A 
3A 00 00 8C
2025-01-31 17:25:12,316 Frame: 0A 70 FF FF 00 3A B2
2025-01-31 17:25:12,332 -------------> COMMAND: F0 F0 0C 60 70 E0 01 24 
01 01 E3 A0
2025-01-31 17:25:12,426 Frame: 0C 60 70 E0 01 24 01 01 E3
2025-01-31 17:25:12,489 Frame: 0A 70 60 E0 80 A1 DB
2025-01-31 17:25:12,819 Frame: 11 70 FF E0 C0 31 C1 04 12 7E 7E 66 00 8A
2025-01-31 17:25:17,546 Frame: 0A 70 FF FF 00 3A B2

> I would just record the power on/ off
> and the send it to check if tx does work.
> 
> Or water on/off you previously decoded.
> 
> [2].
[... 106 lines cut ...]
target=3D"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You are=
 receiving this because you were mentioned.<img src=3D"cid:1738340872679cfa=
=3D"color: transparent; font-size: 0; display: none; visibility: hidden; ov=
erflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height: =
/2625725631</span><span>@</span><span>github</span><span>.</span><span>com&=
gt;</span></span></p>
</div>
</blockquote>
</body></html>

--=_6f0a8b2a094bf1cc6da6f699e80ffbb5
Content-Transfer-Encoding: base64
Content-ID: ***@***.***>
Content-Type: image/gif;
 name=blocked.gif
Content-Disposition: inline;
 filename=blocked.gif;
 size=118

R0lGODlhZAAyAIAAAPrOzgAAACH5BAAAAAAALAAAAABkADIAAAJNhI+py+0Po5y02ouz3rz7D4bi
SJbmiabqyrbuC8fyTNf2jef6zvf+DwwKh8Si8YhMKpfMpvMJjUqn1Kr1is1qt9yu9wsOi8fksvls
KwAAOw==
--=_6f0a8b2a094bf1cc6da6f699e80ffbb5--

--=_6133f0b714520429da548efa1e1aa678--

issalig, 2025-01-31:
Great, you did it!

It would be nice if you share the python script and document your solution.
If you do not feel like creating a full github project I could add it to mine as a readme or under a misc folder.

Titoflegit, 2025-01-31:
As an Electronic Engineer I know that there is a step, sometimes huge, between something that seems to work and something that works really!
  
🤠

Titoflegit, 2025-02-02:
WE update
  
I made some current measurements on the AB bus. Power supply impedance is around 20ohms so to modulate the signal with around 1V as I see we need to draw 50mA.
  
And if you add some perpherals that the AB bus power, the voltage will drop according consumption.
  
So we can not work with a fixed threshold like zeners, its why the levels are different between you and me.   
  
I have designed a circuit with a comparator for Rx, and a fixed current sink source for Tx.
  
Self supplied on AB bus. I will use a Rpi pico W because I am familiar to this micro. If you want, I can plan a second shape for an ESP32 when I will do the PCB layout.
  
Idea is that it is always better to have a solution tested in different system configurations.
  
Let me know.

issalig, 2025-02-02:
Great, my electronics knowledge is limited so having a design that works in different environments is better.
I was also trying to power the ESP8266 from the AB bus but did not work, I think it is because it draws 200mA when connecting to access point and it also corrupted the frames, probably because level voltage went down and the zener trick did not work.

I would like to see how your circuit looks like (and yes I can try it on my AC) . Also rpi pico w sounds interesting.

dzungpv, 2025-04-15:
@Titoflegit can you share the schematics?