issalig/toshiba_air_cond #29: I'm interested in PCB

krishy2, 2023-05-18:
Hello,

I own a Toshiba Estia heat pump. I'd like to assume it has the same "AB" bus. Do you have any experience? Otherwise I'd give a try, just for fun.

Thanks!

issalig, 2023-05-19:
Hi, I do not know Estia system, but it would be nice if you can explore it.
I can send you one soldered pcb with all the components for 8.5 EUR. Bare pcb would be 2 EUR.

Postage to Europe would be 2 EUR for untracked.

Tell me which option do you prefer and the destination country.

alsig, 2023-05-19:
Hey.

 I also have a Toshiba Estia heatpump. 

Do you have more mounted PCBs in stock? Destination country: Denmark 

/Jens

issalig, 2023-05-19:
Yes. I have the components, so I can mount them in a couple of days.

Also, take into account that I do not know if my PCB is compatible with Estia.

krishy2, 2023-05-19:
Hey,
Country: HU
Details: kamuzokrishy at gmail

I understand that you are not sure if it works with Estia, but we will figure it out.

alsig, 2023-05-20:
Super!

Country: Denmark
Details: alsig [at] alsig [dot] dk

issalig, 2023-05-22:
PM sent.

krishy2, 2023-05-22:
Hi,
Something went wrong, I have nothing in my mailbox :(

alsig, 2023-06-04:
Hi, 

Just a bit of feedback.

I've received the PCB two days ago, installed right away and the PCB is working with an Esti heat pump! Great work!

I haven't verified the software yet only the hardware.

I used ESPhome with the following recipe:
```
esphome:
  name: toshiba-link
  friendly_name: toshiba_link

esp8266:
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "magic key :)"

ota:
  password: "top secret password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Toshiba-Link Fallback Hotspot"
    password: "Yet another password"

captive_portal:

uart:
  tx_pin: D8
  rx_pin: D7
  baud_rate: 2400
  parity: EVEN
  debug:
    direction: RX
    dummy_receiver: true
    after:      
      timeout: 100ms
    sequence:
      - lambda: UARTDebug::log_hex(uart::UART_DIRECTION_RX, bytes, ':');    
```     

So @issalig thank you!
/alsig

krishy2, 2023-06-04:
I am curious about it!
My office sent me the message that we received it, but I’m working in Africa, so I’ll be at home in 2 weeks.

Please keep me updated!
I’ll try to integrate as well.

Do you have any plan to have a web interface or smartphone app also?

Thanks,
Kris

Tóth Kristóf
Ügyvezető
Tel.: +36-30-2218678
Control Pro Kft.
Irányítástechnikai Tervező és Kivitelező Kft.
1021 Budapest, Ötvös János u. 3.

issalig, 2023-06-05:
Nice, I have not tried ESPHome, in fact I did not know someone had added support for toshiba. I will take a look.

From my side, I have a web interface that resides in the eps8266.
https://github.com/issalig/toshiba_air_cond#software-installation
and it can be found at
https://github.com/issalig/toshiba_air_cond/tree/master/air/data

alsig, 2023-06-05:
Well there isn't support in esphome, yet. 
All i did was to make a datalogger in esphome. 
I'm hoping that I'll be able to add support to esphome for my heat pump. 
/alsig

issalig, 2023-06-05:
Feel free to open an issue to report your progress with esphome. Then I could add a small guide in the main page.

krishy2, 2023-06-05:
Do we know the datagrams of Estia? Or maybe they are the same as other Toshiba products?

alsig, 2023-06-05:
What I can see from the data is the protocol isn't quite the same. There are similarities but the source and destination adresses are different, the checksum calculation doesn't match your implantation and the temperature conversion are also using a different equation...

issalig, 2023-06-05:
It would be great if you manage to guess the protocol. If you have doubts, just paste the bytes and I can try to guess what´s going on. I tried to document it how I decoded it (main page of repo) but maybe it is not as clear as I though.

You could just ignore crc (at least for RX) . Also you can try https://www.scadacore.com/tools/programming-calculators/online-checksum-calculator/

alsig, 2023-06-05:
Good evening. 

I have had some success last night and tonight. 

It looks like they have modified the protocol a bit. Here is what I have learned so far.

It looks like the remote control has address 0xA0 and the master has address 0x00. 
The status command is 0x1C and ping/alive is 0x10. All (source, destination, opcode-1 and length) are 8 bits. 
A padding(?) bytes have been inserted in position 4 (zero indexed) and the read/write byte is in position 6.
The checksum is actually a crc16 (info here: https://crccalc.com/?crc=A0001C0F00080000FE03C6C30010845C660000&method=CRC-16/MCRF4XX&datatype=hex&outtype=0) 

I also discovered that the temperature calculations are done like this: 
hex_value = temp_deg_c * 2 + 0x20

I have attached a long file for those of you who are curious.

[logs_toshiba-link_logs_with_timestamps.txt](https://github.com/issalig/toshiba_air_cond/files/11657085/logs_toshiba-link_logs_with_timestamps.txt)

issalig, 2023-06-05:
Great, it works for the messages starting with A0 but it does not work the others, for example:
 AF:FB:55:C9:FF:DB:7F:CB:FE:FB:F6:FF:B7: (AF:6F) should be  DC:11

alsig, 2023-06-06:
Damn it :/ Oh well back to the drawing board

issalig, 2023-06-09:
I have tried reveng to guess the CRC algorithm without success https://reveng.sourceforge.io/

issalig, 2023-07-19:
Hi, did you finally manage to decode it?

alsig, 2023-07-19:
Kind of... I did manage to decode part of the protocol but then I got cough up at work and then summer hit... I get back to the project at autumn time. I'll keep you updated...

krishy2, 2023-11-14:
Hello Guys,

Is there any update at your side?

issalig, 2023-11-14:
I wanted to integrate it with homeassistant but I have not tried yet :)

krishy2, 2023-11-14:
But the protocol is clear?

In fact, I’d like to change between two modes in my house at the countryside.

  1.  Most of the times I’m away and I want to have 15C VLT just for frost protection. This would allow the best COP.
  2.  When I’m in the house I need the 45C VLT to ramp it up, later 30-35C is ok.

So, I need to change the VLT continuously.  I haven’t found any possibility to achieve this by digital inputs. Even the modbus converter hasn’t Ethernet or wifi, but costs almost 600€.

If the protocol works, I could issue the VLT set command at least. HA can wait 😊

Thanks

issalig, 2023-11-14:
I was referring to my unit not Estias, so nothing from my side.

krishy2, 2023-11-16:
I’d kill for a Modbus coupler 😊 in that case we could sniff the data.

closed 2024-08-09 by issalig