issalig/toshiba_air_cond #33: Unknown messages

stoffelll, 2023-09-09:
Hi,
I'm using https://github.com/pedobry/esphome_toshiba_suzumi/tree/main and communicating via wifi with my toshiba aircon.

There is a feature integrated to scan for unknown messages/sensors and I got some unknown messages.
I'm now looking for the power consumption of my unit, maybe you have an idea if it is inside of this messages?

This are my messages when the aircon is switched of -> idle mode:
`[07:30:00][I][ToshibaClimateUart:125]: Requesting data from sensor 253, checksum: 183
[07:30:00][I][ToshibaClimateUart:125]: Requesting data from sensor 254, checksum: 182
[07:30:00][W][component:204]: Component api took a long time for an operation (0.98 s).
[07:30:00][W][component:205]: Components should block for at most 20-30ms.
[07:30:00][I][ToshibaClimateUart:271]: Received AC unit power state: OFF
[07:30:00][W][ToshibaClimateUart:291]: Unknown sensor: 129 with value 0
[07:30:00][W][ToshibaClimateUart:215]: Received unknown message with length: 20 and value 02.00.03.90.00.00.0C.01.30.01.00.00.00.05.82.00.01.00.00.A7 (20)
[07:30:00][W][ToshibaClimateUart:215]: Received unknown message with length: 25 and value 02.00.03.90.00.00.11.01.30.01.00.00.00.0A.86.05.00.00.17.00.00.00.00.00.7E (25)
[07:30:00][I][ToshibaClimateUart:263]: Received power select: 100
[07:30:00][W][ToshibaClimateUart:291]: Unknown sensor: 136 with value 66
[07:30:00][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.89.00.00.A5 (18)
[07:30:00][W][component:204]: Component toshiba_suzumi.climate took a long time for an operation (0.08 s).
[07:30:00][W][component:205]: Components should block for at most 20-30ms.
[07:30:01][W][ToshibaClimateUart:291]: Unknown sensor: 144 with value 66
[07:30:01][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.92.00.00.9C (18)
[07:30:02][W][ToshibaClimateUart:291]: Unknown sensor: 148 with value 66
[07:30:02][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.96.00.00.98 (18)
[07:30:02][W][ToshibaClimateUart:215]: Received unknown message with length: 21 and value 02.00.03.90.00.00.0D.01.30.01.00.00.00.06.9A.00.00.00.00.00.8E (21)
[07:30:03][I][ToshibaClimateUart:232]: Received fan mode: Level 5
[07:30:03][I][ToshibaClimateUart:239]: Received swing mode: OFF
[07:30:05][I][ToshibaClimateUart:245]: Received AC mode: COOL
[07:30:05][I][ToshibaClimateUart:220]: Received target temp: 21
[07:30:05][W][ToshibaClimateUart:291]: Unknown sensor: 180 with value 50
[07:30:06][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.B9.00.00.75 (18)
[07:30:06][W][ToshibaClimateUart:291]: Unknown sensor: 186 with value 0
[07:30:06][I][ToshibaClimateUart:252]: Received room temp: 24 °C
[07:30:06][I][ToshibaClimateUart:257]: Received outdoor temp: 16 °C
[07:30:06][W][ToshibaClimateUart:291]: Unknown sensor: 192 with value 0
[07:30:07][W][ToshibaClimateUart:291]: Unknown sensor: 198 with value 0
[07:30:07][W][ToshibaClimateUart:291]: Unknown sensor: 199 with value 16
[07:30:08][W][ToshibaClimateUart:291]: Unknown sensor: 203 with value 16
[07:30:09][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.DB.00.00.53 (18)
[07:30:10][W][ToshibaClimateUart:291]: Unknown sensor: 223 with value 0
[07:30:12][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.EE.2C.00.14 (18)
[07:30:13][I][ToshibaClimateUart:284]: Received special mode: 0
[07:30:13][W][ToshibaClimateUart:215]: Received unknown message with length: 20 and value 02.00.03.90.00.00.0C.01.30.01.00.00.00.05.F8.42.15.36.00.A5 (20)
[07:30:13][W][ToshibaClimateUart:291]: Unknown sensor: 254 with value 48
[07:30:22][I][ToshibaClimateUart:125]: Requesting data from sensor 187, checksum: 249
[07:30:22][I][ToshibaClimateUart:125]: Requesting data from sensor 190, checksum: 246
[07:30:22][I][ToshibaClimateUart:252]: Received room temp: 24 °C
[07:30:22][I][ToshibaClimateUart:257]: Received outdoor temp: 16 °C`

Thanks for your guidance!

stoffelll, 2023-09-09:
This is the log when the aircon is in cooling mode and switched on:

[07:33:56][I][ToshibaClimateUart:271]: Received AC unit power state: ON
[07:33:56][W][ToshibaClimateUart:291]: Unknown sensor: 129 with value 0
[07:33:56][W][ToshibaClimateUart:215]: Received unknown message with length: 20 and value 02.00.03.90.00.00.0C.01.30.01.00.00.00.05.82.00.01.00.00.A7 (20)
[07:33:56][W][ToshibaClimateUart:215]: Received unknown message with length: 25 and value 02.00.03.90.00.00.11.01.30.01.00.00.00.0A.86.05.00.00.17.00.00.00.00.00.7E (25)
[07:33:56][I][ToshibaClimateUart:263]: Received power select: 100
[07:33:56][W][ToshibaClimateUart:291]: Unknown sensor: 136 with value 66
[07:33:56][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.89.00.00.A5 (18)
[07:33:56][W][component:204]: Component toshiba_suzumi.climate took a long time for an operation (0.08 s).
[07:33:56][W][component:205]: Components should block for at most 20-30ms.
[07:33:57][W][ToshibaClimateUart:291]: Unknown sensor: 144 with value 66
[07:33:57][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.92.00.00.9C (18)
[07:33:57][W][ToshibaClimateUart:291]: Unknown sensor: 148 with value 66
[07:33:57][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.96.00.00.98 (18)
[07:33:58][W][ToshibaClimateUart:215]: Received unknown message with length: 21 and value 02.00.03.90.00.00.0D.01.30.01.00.00.00.06.9A.00.00.00.00.00.8E (21)
[07:33:59][I][ToshibaClimateUart:232]: Received fan mode: Level 5
[07:33:59][I][ToshibaClimateUart:239]: Received swing mode: VERTICAL
[07:34:00][I][ToshibaClimateUart:245]: Received AC mode: COOL
[07:34:01][I][ToshibaClimateUart:220]: Received target temp: 21
[07:34:01][W][ToshibaClimateUart:291]: Unknown sensor: 180 with value 50
[07:34:01][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.B9.00.00.75 (18)
[07:34:02][W][ToshibaClimateUart:291]: Unknown sensor: 186 with value 0
[07:34:02][I][ToshibaClimateUart:252]: Received room temp: 24 °C
[07:34:02][I][ToshibaClimateUart:257]: Received outdoor temp: 17 °C
[07:34:02][W][ToshibaClimateUart:291]: Unknown sensor: 192 with value 0
[07:34:03][W][ToshibaClimateUart:291]: Unknown sensor: 198 with value 0
[07:34:03][W][ToshibaClimateUart:291]: Unknown sensor: 199 with value 16
[07:34:03][W][ToshibaClimateUart:291]: Unknown sensor: 203 with value 16
[07:34:05][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.DB.00.00.53 (18)
[07:34:06][W][ToshibaClimateUart:291]: Unknown sensor: 223 with value 0
[07:34:07][W][ToshibaClimateUart:215]: Received unknown message with length: 18 and value 02.00.03.90.00.00.0A.01.30.01.00.00.00.03.EE.2C.00.14 (18)
[07:34:08][I][ToshibaClimateUart:284]: Received special mode: 0
[07:34:08][W][ToshibaClimateUart:215]: Received unknown message with length: 20 and value 02.00.03.90.00.00.0C.01.30.01.00.00.00.05.F8.42.15.36.00.A5 (20)
[07:34:09][W][ToshibaClimateUart:291]: Unknown sensor: 254 with value 48

issalig, 2023-09-09:
My AC looks different (it had no wifi, only wired interface) 
In my AC I need to query a sensor with the given id
https://github.com/issalig/toshiba_air_cond/blob/09efcf16fcfe43d4e7dd11c5d2141aaf0082e5ac/air/toshiba_serial.hpp#L825
Then, read the sensor status https://github.com/issalig/toshiba_air_cond/blob/09efcf16fcfe43d4e7dd11c5d2141aaf0082e5ac/air/toshiba_serial.hpp#L394

In particular in my system current sensor id is 0x6A  https://github.com/issalig/toshiba_air_cond/blob/09efcf16fcfe43d4e7dd11c5d2141aaf0082e5ac/air/toshiba_serial.h#L65

Hope this helps!