issalig/toshiba_air_cond #39: Interested in 3 x pcb's if possible?

prowebuk, 2023-11-21:


iti-praia, 2023-11-21:
Hi, I'm glad you are interested in the project.
I can send you one soldered pcb with all the components for 8.5 EUR. Bare pcb would be 2 EUR.

Postage to Europe would be 3 EUR for untracked, 8 EUR for tracked.

Tell me which option do you prefer and the destination country.

issalig, 2023-11-21:
Hi, I'm glad you are interested in the project.
I can send you one soldered pcb with all the components for 8.5 EUR. Bare pcb would be 2 EUR.

Postage to Europe would be 3 EUR for untracked, 8 EUR for tracked.

Tell me which option do you prefer and the destination country.

prowebuk, 2023-11-22:
Hi,

Can I order 3 x soldered pcb with all the components for 8.5 EUR = 25.50 Euro plus tracked postage @ 8 Euro

My PayPal address is ***@***.******@***.***> if you want to send an invoice?

Address is:

Richard Drage
Proweb (UK) Ltd
Rutland Suite
St James Business Park
Radcliffe on Trent
Nottingham
NG12 2JP
UK

Many thanks

Richard

issalig, 2023-11-22:
Ok, give me your email address and I will MP you.

prowebuk, 2023-11-22:
Paypal is ***@***.******@***.***>, not sure what MP is?

Thanks

Richard

issalig, 2023-11-23:
Sorry, I wanted to say PM (private message).

For some reason I cannot see your address it is shown like this ***@***.******@***.***

Maybe you can try with    your_name AT   your_domain DOT com

prowebuk, 2023-11-23:
😊 my email is richard at proweb dot net

closed 2024-08-09 by issalig