issalig/toshiba_air_cond #51: RBC-AMS55E reporting 01 as master and not 00 as expected

issalig, 2025-01-19:
### Discussed in https://github.com/issalig/toshiba_air_cond/discussions/50

<div type='discussions-op-text'>

<sup>Originally posted by **Gerhard1968** January 19, 2025</sup>
Hello friends, i build the project with the pcb from issalig and connect to my wall controller RBC-AMS55E. I can read the codes on the web side. It shows me the same setting of the RBC-AMS55E on the webside, but its not able to send the commands from webside to the RBC-AMS55E.
Its a different protocol? I see it´s a little differenc in the first and last bit.
And sorry about my bad english!

[image]
[image]
[image]
[image]
</div>

issalig, 2025-01-19:
Interesting, it seems that your master identifies as 01 instead of 00 in my setup
so instead of 00 FE 10 02 80 8A E6 you have **01** FE 10 02 80 8A E7

Maybe changing that line to #define MASTER  0x01 could do the trick. Not sure if MASTER is used in all the commands or I jsut hardcoded 0x00 value (in such a case you can replace it by MASTER)
https://github.com/issalig/toshiba_air_cond/blob/3881bce2d6d0b48492f64b32a3a713a573effbe0/air/toshiba_serial.h#L14

issalig, 2025-01-19:
oooops, I have checked the code and it is always hardcoded, I should change it but in the meantime you can replace all the 0x00 by 0x01

For example power functions should be like this

void air_set_power_on(air_status_t *air) {
  byte data[] = {**REMOTE**, **MASTER**, 0x11, 0x03, 0x08, 0x41, 0x03, 0x18};
  air_send_data(air, data, sizeof(data));
}

void air_set_power_off(air_status_t *air)  {
  byte data[] = {**REMOTE**, **MASTER**, 0x11, 0x03, 0x08, 0x41, 0x02, 0x19};
  air_send_data(air, data, sizeof(data));
}

Gerhard1968, 2025-01-21:
only in toshiba_serial.h ?

issalig, 2025-01-21:
and the functions in toshiba_serial.hpp 

air_set_power_on and the rest. byte 0 is normally source and byte 1 is destination
https://github.com/issalig/toshiba_air_cond/blob/3881bce2d6d0b48492f64b32a3a713a573effbe0/air/toshiba_serial.hpp#L114

Gerhard1968, 2025-01-22:
Sorry i am a hardware type an not software. I change 0x00 to 0x01 but now ich get allways an error:

- I change nothing in the Arduino IDE 2.3.2
- I use a new Wemos D1

In file included from c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:42,
                 from c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\HP EliteDesk\Documents\Arduino\toshiba_air_cond-master\air_copy_20250122164919\air_copy_20250122164919.ino:77:
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp: In instantiation of 'static void ArduinoJson::V730PB22::Converter<T, Enable>::toJson(const T&, ArduinoJson::V730PB22::JsonVariant) [with T = char []; Enable = void]':
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBaseImpl.hpp:139:21:   required from 'bool ArduinoJson::V730PB22::detail::VariantRefBase<TDerived>::doSet(const T&, ArduinoJson::V730PB22::detail::false_type) const [with TConverter = ArduinoJson::V730PB22::Converter<char [], void>; T = char []; TDerived = ArduinoJson::V730PB22::detail::MemberProxy<ArduinoJson::V730PB22::JsonDocument&, ArduinoJson::V730PB22::detail::RamString>; ArduinoJson::V730PB22::detail::false_type = ArduinoJson::V730PB22::detail::integral_constant<bool, false>]'
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:298:29:   required from 'bool ArduinoJson::V730PB22::detail::VariantRefBase<TDerived>::doSet(const T&) const [with TConverter = ArduinoJson::V730PB22::Converter<char [], void>; T = char []; TDerived = ArduinoJson::V730PB22::detail::MemberProxy<ArduinoJson::V730PB22::JsonDocument&, ArduinoJson::V730PB22::detail::RamString>]'
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/VariantRefBase.hpp:82:46:   required from 'bool ArduinoJson::V730PB22::detail::VariantRefBase<TDerived>::set(const T&) const [with T = char []; TDerived = ArduinoJson::V730PB22::detail::MemberProxy<ArduinoJson::V730PB22::JsonDocument&, ArduinoJson::V730PB22::detail::RamString>]'
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/MemberProxy.hpp:38:14:   required from 'ArduinoJson::V730PB22::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >& ArduinoJson::V730PB22::detail::MemberProxy< <template-parameter-1-1>, <template-parameter-1-2> >::operator=(const T&) [with T = char []; TUpstream = ArduinoJson::V730PB22::JsonDocument&; AdaptedString = ArduinoJson::V730PB22::detail::RamString]'
C:\Users\HP EliteDesk\Documents\Arduino\toshiba_air_cond-master\air_copy_20250122164919\process_request.hpp:155:24:   required from here
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:30:18: error: no matching function for call to 'convertToJson(const char [], ArduinoJson::V730PB22::JsonVariant&)'
   30 |     convertToJson(src, dst); // Error here? See https://arduinojson.org/v7/unsupported-set/
      |     ~~~~~~~~~~~~~^~~~~~~~~~
In file included from c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:33,
                 from c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\HP EliteDesk\Documents\Arduino\toshiba_air_cond-master\air_copy_20250122164919\air_copy_20250122164919.ino:77:
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:422:13: note: candidate: 'void ArduinoJson::V730PB22::convertToJson(const ArduinoJson::V730PB22::JsonDocument&, ArduinoJson::V730PB22::JsonVariant)'
  422 | inline void convertToJson(const JsonDocument& src, JsonVariant dst) {
      |             ^~~~~~~~~~~~~
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Document/JsonDocument.hpp:422:47: note:   no known conversion for argument 1 from 'const char []' to 'const ArduinoJson::V730PB22::JsonDocument&'
  422 | inline void convertToJson(const JsonDocument& src, JsonVariant dst) {
      |                           ~~~~~~~~~~~~~~~~~~~~^~~
In file included from c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:42,
                 from c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,
                 from C:\Users\HP EliteDesk\Documents\Arduino\toshiba_air_cond-master\air_copy_20250122164919\air_copy_20250122164919.ino:77:
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:261:13: note: candidate: 'void ArduinoJson::V730PB22::convertToJson(const Printable&, ArduinoJson::V730PB22::JsonVariant)'
  261 | inline void convertToJson(const ::Printable& src, JsonVariant dst) {
      |             ^~~~~~~~~~~~~
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:261:46: note:   no known conversion for argument 1 from 'const char []' to 'const Printable&'
  261 | inline void convertToJson(const ::Printable& src, JsonVariant dst) {
      |                           ~~~~~~~~~~~~~~~~~~~^~~
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:191:56: note: candidate: 'template<class T> ArduinoJson::V730PB22::detail::enable_if_t<ArduinoJson::V730PB22::detail::IsString<TString>::value> ArduinoJ [... cut ...]
c:\Users\HP EliteDesk\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Variant/ConverterImpl.hpp:30:18: note:   mismatched types 'const ArduinoJson::V730PB22::detail::VariantRefBase<TDerived>' and 'const char []'
   30 |     convertToJson(src, dst); // Error here? See https://arduinojson.org/v7/unsupported-set/
      |     ~~~~~~~~~~~~~^~~~~~~~~~
Mehrere Bibliotheken wurden für "SoftwareSerial.h" gefunden
  Benutzt: C:\Users\HP EliteDesk\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\SoftwareSerial
  Nicht benutzt: C:\Users\HP EliteDesk\Documents\Arduino\libraries\EspSoftwareSerial
exit status 1

Compilation error: exit status 1

issalig, 2025-01-23:
Looks like ArduinoJson is complaining.

Probably they have change something in the APIs
I am using ArduinoJson 6.21.3 and it seems you are using v7

Maybe downgrading it will work.

These are the versions I use (might update it some day :) ) 

Usando librería LittleFS con versión 0.1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/LittleFS 
Usando librería ESP8266WiFi con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WiFi 
Usando librería WiFiManager con versión 2.0.16-rc.2 en la carpeta: /home/issalig/Arduino/libraries/WiFiManager 
Usando librería ESP8266WebServer con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266WebServer 
Usando librería DNSServer con versión 1.1.1 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/DNSServer 
Usando librería arduinoWebSockets con versión 2.3.6 en la carpeta: /home/issalig/Arduino/libraries/arduinoWebSockets 
Usando librería ESPAsyncTCP con versión 1.2.2 en la carpeta: /home/issalig/Arduino/libraries/ESPAsyncTCP 
Usando librería SPI con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SPI 
Usando librería Wire con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Wire 
Usando librería Adafruit_GFX_Library con versión 1.11.7 en la carpeta: /home/issalig/Arduino/libraries/Adafruit_GFX_Library 
Usando librería Adafruit_BusIO con versión 1.11.6 en la carpeta: /home/issalig/Arduino/libraries/Adafruit_BusIO 
Usando librería Adafruit_SSD1306 con versión 2.5.7 en la carpeta: /home/issalig/Arduino/libraries/Adafruit_SSD1306 
Usando librería ArduinoOTA con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ArduinoOTA 
Usando librería ESP8266mDNS con versión 1.2 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/ESP8266mDNS 
Usando librería ArduinoJson con versión 6.21.3 en la carpeta: /home/issalig/Arduino/libraries/ArduinoJson 
Usando librería NTPClient con versión 3.2.1 en la carpeta: /home/issalig/Arduino/libraries/NTPClient 
Usando librería SoftwareSerial con versión 6.12.7 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/SoftwareSerial 
Usando librería Ticker con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Ticker 
Usando librería DHT_sensor_library con versión 1.4.4 en la carpeta: /home/issalig/Arduino/libraries/DHT_sensor_library 
Usando librería Adafruit_BMP085_Library con versión 1.2.2 en la carpeta: /home/issalig/Arduino/libraries/Adafruit_BMP085_Library 
Usando librería Hash con versión 1.0 en la carpeta: /home/issalig/.arduino15/packages/esp8266/hardware/esp8266/3.0.2/libraries/Hash 
Usando librería Adafruit_Unified_Sensor con versión 1.1.9 en la carpeta: /home/issalig/Arduino/libraries/Adafruit_Unified_Sensor

Gerhard1968, 2025-01-25:
I downgrade ArduinoJson and the upload ok now, fine thank you. 

I can switch now Fan/Dry/Cool/Head and Low/Med/High/Auto with the webinterface, but ON/OFF dosn´t work!
Mayby this line in toshiba_serial.h is the problem?

42   #define TIMER_HW_CANCEL     0x01

in the toshiba_serial.hpp i wrote

void air_set_power_on(air_status_t *air) {
byte data[] = {REMOTE, MASTER, 0x11, 0x03, 0x08, 0x41, 0x03, 0x18};
air_send_data(air, data, sizeof(data));
}

void air_set_power_off(air_status_t *air) {
byte data[] = {REMOTE, MASTER, 0x11, 0x03, 0x08, 0x41, 0x02, 0x19};
air_send_data(air, data, sizeof(data));
}

Here the log´s:

[image]
[image]

issalig, 2025-01-27:
Nice, there is some progress.

Regarding this line 42 #define TIMER_HW_CANCEL 0x01 , this is not the culprit, this just affect timer functions but that feature is not working well.

For power on you are sending 40 01 11 03 08 41 03 19   which looks ok to me.

Can you press power on/off on the "real" remote and check what is sent in the web (DEBUG CMD)?
Maybe we are missing sth.

Debug RAW is not really important (mainly to test the function that separate frames from byte stream). You can just use DEBUG CMD