issalig/toshiba_air_cond #58: Optimistic behaviour when changing target temp from HA

makusets, 2025-08-05:
Thanks @issalig for this amazing piece of code. I love the new design and neat MQTT discovery.

I am not sure I have identified the issue correctly, and it might actually be a HA issue, but this is what I see happening.

When you change the target temp from the websocket, it seems that you first send the instruction and wait for confirmation, before actually updating what the display shows, This is a neat way of doing it, as often messages seem to get lost and you wouldn't know if the temp was really changed or not.  In the next cycle (30s?) the MQTT topic status temp is published and HA updates accordingly.

However, if I change the temp from HA, and the message gets lost, HA frontend behaves in an optimistic way and doesn't wind back the temperature. The sequence is as follows:
- I change the target temp in HA from 22 to 23, HA thermostat card immediately reflects this
- I can see the message gets published properly as a set temperature in MQTT
- Esp reads the MQTT set temp and send the instruction to the central unit, the websocket page still shows 22 as target temp
- The message gets lost, central unit doesn't acknowlegde? or whathever, the websocket remains the same at 22
- Next status temp from esp (every 30s?) to mqtt still sends 22 (nothing changed).
- HA doesn't doesn't update the displayed target temp in the frontend card, still shows 23 in an optimistic way. (wait over 10 min and still at 23)

I believe the issue might be the frontend card, because the topic gets published correctly and if I later change the target temp to something else through the websocket, it then gets updated in HA frontend.

Finally, if there is an acknowledge message from central unit when target temp is changed, then I think an immediate MQTT message to update HA would be a good idea, 30s seems a bit long. Same might apply to any setting changed I guess. 

Any thoughts?

Thanks again.

issalig, 2025-08-05:
Hi @makusets,  thanks for the analysis.From my expierence it only takes some seconds (<30) to see the update in HA. Once the message is received by mqtt.cpp, statusChanged is true and then status message is sent to websockets in notifyWebsocketCLients()
I am checking the code and maybe it is better to do the notify (and update the air_status variable by calling air_parse_serial)  and then update it in HA. That way the update do not have to wait until the next cycle.
The code would be like this
In mqtt.cpp
 if (statusChanged) { print_log("[MQTT] command processed successfully\n"); notifyWebSocketClients(); // Update status update to websocket clients handleMQTT(); // Call handleMQTT to send updated status to HA }
In process_request.cpp
void notifyWebSocketClients() { String message; //update status air_parse_serial_data(&air_status); yield(); message = air_to_json(&air_status); webSocket.broadcastTXT(message); //reset serial buffers air_status.buffer_cmd = String(""); air_status.buffer_rx = String("");}
Also,  30 secs period may be too much. Lowering it to 10 secs could improve the responsiveness without affecting to the performance.
With these changes I think it should work. Tell me if that solves your problem.

From my side, I want to integrate telegram bot control but it interferes with the links2004/WebSockets library I use (only one SSL certificate is allowed and stoping /starting connections seem not work ok). I am considering to change to  ESPAsyncWebServer with AsyncWebSocket but I had problem some years ago in this project and that is why I chose links2004 library.
BR.

makusets, 2025-08-06:
Since you are only accessing the websocket from within your own wifi, you could use simple ws instead of wss and you will have a free ssl for telegram, am i right? Alternatively, as ESP8266 only handles 1 certificate, you could upgrade to esp32, that would work and would be a more stable board than the d1mini. I don't think starting/stopping would be a good approach, plus you might consume a lot of resources.

Regarding the above issue, the main issue was when updating from HA, but message not being received by the central unit. That's when HA doesn't roll back the dial. Am i right that you wait for an acknowledgement from central unit after a command is sent?

issalig, 2025-08-06:
No, I do not wait for an ack.
I just read the status messages from AC and update status variable which reported to web and mqtt.